google/openhtf #985: Question: What are the benefits of using plugs as opposed to plain python libraries for example?

arjuna-dev, 2021-07-06:
I am starting to use OpenHtf and I am not able to see the clear use of plugs. What are the benefits of using a plug wrapping a python library as opposed to just writing your test functions directly in python and the library?

In my case I am now just writing functions without plugs (python MIDO library) and testing those and it seems to work fine. I did an implementation with plugs and the end results seem to be pretty much the same. It just seemed to make my code more complex. What am I missing?

glados-verma, 2021-07-08:
At the moment, plugs provide two main benefits:

1. tearDown functionality - think of it as a try/finally for the plug over the entire test. Whether the test passes, fails, times out or errors, tearDown will be called on the plug.
2. OpenHTF logger - the `logger` attribute in the plug can be used instead of the `logging` module, the benefit being that the logs go into the OpenHTF logger rather than the root logger (or whatever your libraries might be configured for). This allows better organization of the logging output.

glados-verma, 2022-10-12:
There are a few additional benefits; for example, you can have placeholder plugs in a plugs, that are satisfied with a concrete plug implementation later, allowing you to use the same phase in different tests using different plugs. This is an uncommon requirement though, as far as I can tell.

closed 2022-10-12 by glados-verma