google/openhtf #987: Different results using only python vs. OpenHtf?

arjuna-dev, 2021-08-02:
I am testing the UART protocol o a [Digilent DIgital Discovery](https://store.digilentinc.com/digital-discovery-portable-usb-logic-analyzer-and-digital-pattern-generator/). I am running some tests using UART protocol to both send and receive in the device. 

I am getting very different results from running only the python functions I am using as tests as opposed to letting OpenHtf use those functions.

For example, for a function measuring how many packets per second I can send the result in OpenHtf is around 1272 whereas if I run the python function I get a value of around 170. For detecting the fastest sent message in a period of time python gives me a value of 0.00425 whereas OpenHtf 6.32e-07. 

This are my tests:
```
@htf.PhaseOptions(name='DDD_Tests')
@htf.measures(htf.Measurement("DDD_UART_Speed_One_Message_Per_Packet")
              .doc("DDD UART max speed for sending one MIDI message data per packet")
              .in_range(0, 0.07))
@htf.measures(htf.Measurement("DDD_UART_Speed_Multi_Messages_Per_Packet")
              .doc("DDD UART max speed for sending multiple MIDI message data per packet")
              .in_range(0, 0.07))
@htf.measures(htf.Measurement("mbps_UART_DDD")
              .doc("DDD UART mbps")
              .in_range(1000, 10000000000))
@htf.measures(htf.Measurement("Packets_Per_Second_UART_DDD")
              .doc("DDD UART packets per second")
              .in_range(1000, 10000000000))
def DDDTests(test):
    test.measurements.DDD_UART_Speed_One_Message_Per_Packet = onlySendOneMessageRecord(1)
    test.measurements.DDD_UART_Speed_Multi_Messages_Per_Packet = onlySendMessagesRecord(1, 32)
    test.measurements.mbps_UART_DDD = mbps(1, 32)
    test.measurements.Packets_Per_Second_UART_DDD = mbps(1, 32)

test = htf.Test(
    htf.PhaseGroup(setup=[], main=[DDDTests]),
)

test.add_output_callbacks(Measurements())
test.add_output_callbacks(json_factory.OutputToJSON('./ddd/results/results.{start_time_millis}.json', indent=4))
test.execute()
````

And the python functions:

```
def onlySendOneMessageRecord(testSpan):
    TX = messageCreator()
    minElepasedTime = 1
    t0 = time.perf_counter()
    while t0 + testSpan > time.perf_counter():
        t1 = time.perf_counter()
        dwf.FDwfDigitalUartTx(hdwf, TX, c_int(sizeof(mystruct)))
        elapsedTime = time.perf_counter() - t1
        if elapsedTime < minElepasedTime:
            minElepasedTime = elapsedTime
    return minElepasedTime

def onlySendMessagesRecord(testSpan, numberOfMessages):
    TX = multiMessageCreator(numberOfMessages)
    minElepasedTime = 1
    t0 = time.perf_counter()
    while t0 + testSpan > time.perf_counter():
        t1 = time.perf_counter()
        dwf.FDwfDigitalUartTx(hdwf, TX, c_int(sizeof(mystruct)*numberOfMessages))
        elapsedTime = time.perf_counter() - t1
        if elapsedTime < minElepasedTime:
            minElepasedTime = elapsedTime
    return minElepasedTime

def mbps(testSpan, numberOfMessages):
    TX = multiMessageCreator(numberOfMessages)
    minElepasedTime = 1
    t0 = time.perf_counter()
    counter = 0
    while t0 + testSpan > time.perf_counter():
        t1 = time.perf_counter()
        dwf.FDwfDigitalUartTx(hdwf, TX, c_int(sizeof(mystruct)*numberOfMessages))
        elapsedTime = time.perf_counter() - t1
        counter += 1
    return counter*sys.getsizeof(TX)*8/testSpan/1000000

def packetsPerSecond(testSpan, numberOfMessages):
    TX = multiMessageCreator(numberOfMessages)
    minElepasedTime = 1
    t0 = time.perf_counter()
    counter = 0
    while t0 + testSpan > time.perf_counter():
        t1 = time.perf_counter()
        dwf.FDwfDigitalUartTx(hdwf, TX, c_int(sizeof(mystruct)))
        elapsedTime = time.perf_counter() - t1
        counter += 1
    return counter/testSpan
```

I must be missing something here.

jettisonjoe, 2021-11-21:
My first thought is maybe there's a difference between how you're initializing the test equipment when you're running the OpenHTF test versus calling the function manually? Whatever the difference is, I don't think it's in the code you've pasted here. Calling a library function from an OpenHTF test, conceptually, should be no different from calling a library function in your own script.

glados-verma, 2022-10-12:
Please reopen with a complete repro if you're still seeing issues.

closed 2022-10-12 by glados-verma