google/openhtf #1001: Support tornado>=6.0

wallacbe, 2022-04-04:
Support tornado>=6.0

dbhatman, 2022-10-12:
FYI #934 includes some failure example logs

glados-verma, 2022-10-14:
When trying to run the dashboard_server with Python3.10, I see this error: `AttributeError: module 'collections' has no attribute 'MutableMapping'`. The current version of Tornado therefore caps us to using Python3.9 at the latest.

matt-donahoe-skydio, 2022-10-22:
I added explicit asyncio event loops in the `run()` method of any `threading.Thread` subclass, like `WebGuiServer` and `StationWatcher`.

```
asyncio.set_event_loop(asyncio.new_event_loop())
```

I also had to delay the call to `self.server.add_sockets(sockets)` until the event loop was set for the thread, otherwise tornado would grab a reference to the eventloop of the main thread and get blocked, never responding to requests.

Seems to work so far, but I'm very new to OpenHTF, so I doubt I caught everything. I'll post a PR in a few days if no one beats me to it.

closed 2024-10-30 by glados-verma