google/openhtf #1017: is this project dead ?

mbenabda, 2022-08-08:
Hi there,

Is someone still maintaining this project ?

I've seen commits these past few months, and interesting ideas (diagnoses) but as there has been no release since 2018, and a bunch of dangling PRs, i am wondering if i should still rely on it, or consider making any PRs.

Any answer appreciated, thank you !

glados-verma, 2022-08-26:
Hello, this project isn't dead, it's pretty stable at the moment, and we push updates infrequently. However we haven't had time to go through issues and pull requests of late; I'll try to give them a look soon.

Regarding releases - do you mean a Pypi release?

mbenabda, 2022-08-29:
Hello @glados-verma thank you for your answer !

would you be able to share an approximative date at which you expect to be starting looking at issues and PRs ?

maybe you could also let us know how we can help. i think i am not the only one getting tremendous value from this project, and that is willing to help. maybe if you could explain how governance works on this project ?

regarding release, i was referring to both github and pypi releases

thank you again

glados-verma, 2022-08-29:
Let me get back to you on Friday about these points, I need to discuss them with my team first.

jonasclaes, 2022-08-29:
In reply to @mbenabda , I have recently setup this framework for use at a company in the Netherlands, producing parts for ASML. I would love to see more features and collaborate on these.

lalten, 2022-09-15:
Hey @cricdecyan I just want to say I'm really happy that you guys bring some life into the project with merging PRs! It will be great if we can continue like this :muscle:

cricdecyan, 2022-09-15:
@lalten you are welcome. We are trying to bring in as many PRs as possible, some are easy to bring in, others less.

umamibeef, 2022-09-26:
Is there an expected timeline for a PyPI release?

mitchell-ostler-skydio, 2022-09-27:
> Is there an expected timeline for a PyPI release?

I would also be curious to know about this

glados-verma, 2022-09-29:
I'll try to make a release this week to PyPI.

Closing this issue; I started the first Discussion for this project, please take a look!

closed 2022-09-29 by glados-verma

glados-verma, 2022-10-05:
Release is pushed out a bit as we identified a few changes we'd like to include in the release. Current plan is to make the release next week.