google/openhtf #1019: Request: visualize FAIL_AND_CONTINUE as red color on the web frontend.

antonysigma, 2022-08-18:
@fahhem and @arsharma1 A follow-up question: if I enable the web frontend, will it show a green color (indicating a PASS) or a red color?

I can confirm that `OutputToJSON` returns the `FAIL_AND_CONTINUE` state, but not sure how the web frontend will behave.

_Originally posted by @antonysigma in https://github.com/google/openhtf/issues/616#issuecomment-1219983021_

The context: I wish to return `htf.PhaseResult.FAIL_AND_CONTINUE` without any `@htf.measures()` decorators in a test phase.

The bug: `OutputToJSON` returns the `FAIL_AND_CONTINUE` state, but the web frontend is still showing a green color of the test, indicating a false positive.

lalten, 2022-09-01:
Try https://github.com/google/openhtf/pull/1020, that probably does what you want :)

dbhatman, 2022-10-12:
Yes, after lalten@'s change this should generally work when you enable the Web GUI! See #1020 which has a quick screen shot though not specifically of the red with FAIL but that is the expected behavior! We should also probably include some screen shots as part of some user docs (i.e. #44).

Going to close this out. Please reopen or create a new issue if you have more followups!

closed 2022-10-12 by dbhatman