google/openhtf #1021: AttributeError: '_Configuration' object has no attribute 'KeyAlreadyDeclaredError'

lalten, 2022-09-02:
https://github.com/google/openhtf/commit/c7ab5080ed64c9b81721eb4de72bba2bdf6b83cc#diff-53a08cb5972204052f624d14da73b125480659e3cad641e693ebc4debe580a0c Introduces a regression where this won't work anymore:
```py
from openhtf.util import conf
conf.KeyAlreadyDeclaredError
```
```
File ~/.local/lib/python3.8/site-packages/openhtf/util/configuration.py:411, in _Configuration.__getattr__(self, field)
    409   return self[field]
    410 # Config keys all begin with a lowercase letter, so treat this normally.
--> 411 raise AttributeError("'%s' object has no attribute '%s'" %
    412                      (type(self).__name__, field))

AttributeError: '_Configuration' object has no attribute 'KeyAlreadyDeclaredError'
```
This is unfortunate because existing code imports that for use in try/except blocks.

dieppedalle, 2022-09-03:
Hi @lalten,

In #1004 we have moved the `configuration` to its own module instead of living in `conf`. This pattern is consistent with other libraries that expose a global "singleton" instance in their APIs such as [`absl.flags`](https://abseil.io/docs/python/guides/flags). As you may have seen, `from openhtf.util import conf` is now marked as a legacy library. Could you attempt to use the below snippet instead?
```python
from openhtf.util import configuration
configuration.KeyAlreadyDeclaredError
```

Thanks!

Gauthier

lalten, 2022-09-03:
That works, and is what I ended up doing. It's just unfortunate that the change is not fully backwards-compatible and I need to modify existing application code to be able to use the most recent OpenHTF.
Probably one could add some more getattr magic to the conf module to provide a fully backwards-compatible solution?

lalten, 2022-09-05:
Closing the issue as I guess the behaviour is intentional ;)

closed 2022-09-05 by lalten

dbhatman, 2022-09-07:
Yes this was intentional. For now conf.py continues to exist to make life easier for users (ie basic conf behaviors are not changed) but it is not fully backwards compatible. Apologies for the breaking change without more clear notice. FYI, In a future update (requires cleanup even inside this codebase) we will likely delete conf.py and using configuration.CONF will be necessary instead.

Some more background:

"from openhtf import conf" is not importing the actual conf.py module but instead some magic is being done to point to an actual _Configuration object. This was also the case previously, but the _Configuration object used to contain all the exceptions and even had some magic to contain imported modules (basically it was an overly complicated singleton).

The magic was breaking static analyzers (i.e. mypy, pytype) and made improving conf more painful.