google/openhtf #1039: web_gui build is broken

lalten, 2022-09-20:
As pointed out in https://github.com/google/openhtf/pull/1036, the web_gui does not currently build.

The web gui dependencies really should get some attention, they are severely outdated.
One big item is `node-sass`, which is pinned at 4.8.3:
https://github.com/google/openhtf/blob/e7241739d0d818588a05a68742ca16402a4152f6/openhtf/output/web_gui/package.json#L73
Node-sass 4.8.3 supports only up to Node 9 (https://github.com/sass/node-sass/releases/tag/v4.8.3), and indeed building it during `npm install` will fail with a newer Node. Luckily it's easy to switch node versions with `n`

However even with node-sass built, the web gui is broken on master.
`npm run build` fails with:
```
ERROR in js/vendor.6e12e27800e23a08245f.js from UglifyJs
Unexpected token name «key», expected punc «;» [./~/@angular/core/fesm2015/core.js:18,0][js/vendor.6e12e27800e23a08245f.js:1051,13]

ERROR in [at-loader] ./src/app/plugs/user-input-plug.component.ts:46:27 
    TS2304: Cannot find name 'SafeHtml'.
```

The SafeHtml error is easily fixed by cherry-picking https://github.com/google/openhtf/pull/1035

The Uglify.js error is a little harder. The problem seems to be that it does not support ES6 (https://stackoverflow.com/a/59156371/5559867), which is curious because tsc will still compile to ES5:
https://github.com/google/openhtf/blob/e7241739d0d818588a05a68742ca16402a4152f6/openhtf/output/web_gui/tsconfig.json#L3

It turns out that
```
❯ npx rimraf dist && npx webpack --progress --profile --bail
```
seems to work, while `npm run build` (which should invoke the same, see below) fails.
https://github.com/google/openhtf/blob/e7241739d0d818588a05a68742ca16402a4152f6/openhtf/output/web_gui/package.json#L5

There are a few more problems with the scripts in the package.json.
E.g. there is some protractor e2e stuff, but no protractor config file, so this won't work.

Same with `karma`, at first `npm run test` will complain about
```
Error: yargs parser supports a minimum Node.js version of 10. Read our version support policy: https://github.com/yargs/yargs-parser#supported-nodejs-versions
```
Remember, we have to use Node 9 for node-sass. But anyways, let's temporarily upgrade to Node 10. Karma starts:
```
20 09 2022 21:59:52.105:INFO [karma-server]: Karma v6.4.0 server started at http://localhost:9876/
```
And immediately fails with
```
Karma v 6.4.0 - connected; test: karma_error You need to include some adapter that implements __karma__.start method!;
```
which happens because it turns out there is no karma config file either.

In a nutshell, the state of the webinterface build is a mess and really needs some love.

lalten, 2022-09-21:
It looks like running webpack directly works because the failing uglify js is only called when `isProd` is true, which is not the case when invoking directly instead of via `npm run build`.
https://github.com/google/openhtf/blob/master/openhtf/output/web_gui/webpack.config.js#L19

lalten, 2022-09-21:
OK it turns out if you also revert https://github.com/google/openhtf/pull/1014 and the change in `openhtf/output/web_gui/src/app/stations/station/station.component.spec.ts` from https://github.com/google/openhtf/pull/1004 the build works again.

lalten, 2022-09-21:
I think it would make a lot of sense to add a CI check that ensures `npm run build` always works.

glados-verma, 2022-10-11:
Currently the node version needs to be downgraded to 9 in order to successfully build. #1024 might help with that, I will take a look.

closed 2022-10-11 by glados-verma