google/openhtf #1067: frontend_example with PyInstaller

1am, 2022-10-21:
Hello. 

I'm just asking since I'm curious about one aspect of the project. While trying to make a standalone binary from the [frontend_example.py](https://github.com/google/openhtf/blob/master/examples/frontend_example.py), it launches but I end up with a "500: Internal Server Error" message in the browser. This is it, no error in terminal. I assume that the binary isn't bundling the correct resources or that the path is different than when running from local script like in SO thread [Bundling data files with PyInstaller (--onefile)](https://stackoverflow.com/a/44352931). 

If the above assumption would be correct, most likely I'd have to figure out a way to swizzle the path to [openhtf/openhtf/output/web_gui/dist/](https://github.com/google/openhtf/tree/master/openhtf/output/web_gui/dist) folder so that the server serves these contents or something else would be necessary?

Best regards,

1am, 2022-11-10:
In the end it's actually quite easy, just add this to `datas`

```
('../.venv/src/openhtf/openhtf/output/web_gui','./openhtf/output/web_gui')
```

You obviously will need to adjust the "../.venv/src/openhtf/openhtf/output/web_gui" to appropriate path for your dev env setup.

glados-verma, 2022-11-10:
Thanks for reporting and identifying the issue! I'm reopening so that we can make this example work out of the box.

kzhuo333, 2023-05-17:
Adding more detail for those who have trouble following the above discussion.
I found two ways to do this for my setup.
In my workfolder, I want to create an executable from a file called "main.py" and I have a ".venv" virtual environment folder containing the "web_gui" subfolder.

Method 1 using VSCode terminal:
```
pyinstaller --onefile --add-data .venv/Lib/site-packages/openhtf/output/web_gui;./openhtf/output/web_gui main.py
```

Method 2 in the form of a python script:
```
import PyInstaller.__main__

PyInstaller.__main__.run([
        'main.py',
        '--onefile',
        '--add-data',
        '.venv/Lib/site-packages/openhtf/output/web_gui;./openhtf/output/web_gui'
    ])
```