google/openhtf #1080: How to build a test running multiple test groups concurrently, but showing I/O data of all test groups on the same webpage?

daviddengma, 2023-01-27:
We are building a manufacturing test station, which needs to run test groups over 4 modules at a time. Those 4 test groups share couple of instruments so they will need to take turn on those test cases; but other than that they have their own resources for remaining test cases. All the I/O data, like user input, prompts and logs, test results, of 4 test groups need to be displayed and refreshed on a single webpage. How does openhtf support such a use case? Any suggestions are highly appreciated.

kehrazy, 2023-04-10:
wouldn't measurements work? 4 monitors keep updating ur instruments, handling ur resources, etc, etc.
```python
def handle_module_1(test, module_api):
   test.measurements.module_1 = module_api.get_measurement()

@openhtf.measures(
    openhtf.Measurement('module_1')
)
@openhtf.monitors(handle_module_1)
def your_test_phase(test: openhtf.TestApi):
    test.get_measurements("module_1")  # this gets the data from all the other phases
```

something like that?

kehrazy, 2023-05-16:
well, similar functionality is already in the _MonitorThread class, however:

- if we use any sort of input that requires human intervention, the whole program locks up. input blocks globally (does it?)

- we should make a "ConcurrentGroup", which should (should it?) open up a whole bunch of threads...

```python

if __name__ == "__main__":
    executor = htf.Test(
        htf.ConcurrentGroup(
            ...  # < a list of phases, just as a thread
        ),
        htf.ConcurrentThread(
            ...
        )
    )
    executor.execute()
```

So, the executor has a global threading.Event(). When all of the ConcurrentThreads stop -> kill the runner thread?

This way this API wouldn't break the existing APIs.

Although I wouldn't have any idea how does this mess up the GUI. I would imagine that the GUI would need some reimagination.