google/openhtf #1089: HIL testing?

kehrazy, 2023-03-31:
how would i go about making a test that goes on for n amount of time? i need to

- gather data
- check the data
- log the erroneous measurements, if needed
- repeat, if the time has not passed

does this library has this functionality? thanks

glados-verma, 2023-04-05:
I don't see a direct way to specify that a phase should be repeated until `x` time has passed (you can set a timeout, but that's not exactly what you're asking about here). I'd suggest setting a measurement that records the start time, and then running a phase on repeat until the desired time has passed. In that phase you can check this measurement and return `PhaseResult.REPEAT` until the desired amount of time has passed.

kehrazy, 2023-04-05:
Hey, thanks for the response. Yeah, this kind of functionality might be out of the design goals of the library, however, something along the lines of

```python
#: runs for 60 secods, with a .5 sleep in between the runs
@htf.PhaseOptions(looping=True, interval=.5, duration=60)
def polling_test():
    pass

def main():
    test = htf.Test(
        #: this behaviour should spawn a threading.Thread instead,
        #: and skip right into the next phase
        polling_test,
        some_other_phase
    )
    test.execute()

if __name__ == '__main__':
    main()
```

would be extremely useful.

glados-verma, 2023-04-06:
Agreed it would be useful! One question I have is how do you make a phase run for exactly the amount of time specified "from outside the phase" i.e. from the framework. E.g. in your example, is `polling_test` run in a thread for 60 seconds, at the end of which the thread is terminated and a new one is started?

I suppose your use case could be met by adding a `repeat_duration` (perhaps with a better name) that works like `repeat_limit` but instead measures total time rather than number of repeats.

kehrazy, 2023-04-07:
In my concrete example, I imagined the `polling_test` runs the phase body while the `timeout` has not been reached yet, the thread sleeps for the specified `interval`.

So, I would suppose it would run something along the lines of:

- measure current time
- time passed? if yes, stop/return from the thread.

kehrazy, 2023-04-08:
if anyone is wondering - you should look into the "core.monitors" file.