google/openhtf #1096: KeyError in `PlugManager.provide_plugs`

kehrazy, 2023-04-20:
```
    def provide_plugs(
        self, plug_name_map: Iterable[Tuple[Text, Type[base_plugs.BasePlug]]]
    ) -> Dict[Text, base_plugs.BasePlug]:
        """Provide the requested plugs [(name, type),] as {name: plug instance}."""
        return {name: self._plugs_by_type[cls] for name, cls in plug_name_map}
```

raises 
> KeyError: <class 'plugs.`plug_name`'>.

Any help?

glados-verma, 2023-04-20:
What does your phase description looks like? Have you used `with_plugs` with the same keyword as in the phase function?

kehrazy, 2023-04-22:
Correct me if I'm wrong, by "phase description" you mean something like:

```python
@htf.monitors('measurement_one', monitor_one)
@htf.monitors('measurement_two', monitor_two)
@htf.plugs(some_plug=SomePlug)
def simple_phase(test: htf.TestApi, some_plug: SomePlug):
    ...
```

This raises a KeyError. However, this is not a stable error - I can not seem to get a stable reproducible example.

kehrazy, 2023-04-22:
If I reverse the `@monitors` order, like this:

```python
@htf.monitors('measurement_two', monitor_two)
@htf.monitors('measurement_one', monitor_one)
@htf.plugs(some_plug=SomePlug)
def simple_phase(test: htf.TestApi, some_plug: SomePlug):
    ...
```

then the second `@monitors` decorator raises the error.

kehrazy, 2023-04-22:
Pretty sure I've found the issue.

The `@monitors` decorator uses a `@plugs.plug` in the `monitored_phase_func` wrapper.

When running the unmodified code, the behaviour is something like:
```
[ for 'monitor_one': new plug -> result -> "update_kwargs" ] -> False
[ for 'monitor_two': new plug -> result -> "update_kwargs" ] -> False

hello from monitor one!

Exception in thread MonitorTwoThread:
   ...
```

which for some reason makes the `TestExecutor` ignore the required plugs.

If I force the `update_kwargs` to True, the TestExecutor runs just fine, as expected. 
```
[ for 'monitor_one': new plug -> result -> "update_kwargs" ] -> True
[ for 'monitor_two': new plug -> result -> "update_kwargs" ] -> True

hello from monitor one!
hello from monitor two!
```
Is this intended?

kehrazy, 2023-05-11:
Should be closed with #1099.

closed 2023-05-11 by kehrazy

marcovannoord, 2023-05-11:
I can confirm that it works again now, thanks!