google/openhtf #1103: Progress bar/completion status stops if any of the tests are marked as FAIL.

Monoloqu3, 2023-06-02:
It's a bit confusing that the progress bar/completion status is stopped even though the test suite is still running. 
I'm not sure if this is a bug or it was planed to be like that.
Has anyone found a workaround for this?

kehrazy, 2023-06-04:
https://github.com/google/openhtf/blob/5f11f03af022f7d041d14f135512ca9e71d8359b/openhtf/output/web_gui/src/app/stations/station/test-summary.component.ts#L44-L48

Looks like this is intended behaviour.

I assume the logic here is that we can easily look into our GUI and find out easily when did our tests fail.

Monoloqu3, 2023-06-05:
> If I understand correctly, this is a bug as it looks like the total count of run phases is incorrect?
> 
> From a quick look https://github.com/google/openhtf/blame/5f11f03af022f7d041d14f135512ca9e71d8359b/openhtf/output/web_gui/src/app/stations/station/test-summary.component.ts#L47 should change from `PhaseStatus.fail` to `PhaseStatus.waiting` as fail is a completed phase indicator. If you have a chance please give that a shot and feel free to send a PR. Otherwise I'll see if I can find some time to test this out.

Thanks! I will this soltution and let you know about the result.

Monoloqu3, 2023-06-08:
> If I understand correctly, this is a bug as it looks like the total count of run phases is incorrect?
>
> From a quick look [...]

Could be a nooby issue, but it seems like in web_gui folder I don't have "src" folder. I installed OpenHTF using pip.
This is how my path looks like: C:\Users\name\venv\openhtf\Lib\site-packages\openhtf\output\web_gui

[image]