google/openhtf #1104: How could I call a PhaseDescriptor explicitly?

kehrazy, 2023-06-04:
Is this possible?
```python
@htf.plugs(stuff=SomePlug)
def cool_phase(api: htf.TestApi, stuff: SomePlug):
    return stuff.do_stuff() == StuffResult.SUCCESS

def main_phase(api: htf.TestApi, stuff: SomePlug):
    # What can I do to call cool_phase here?
    
if __name__ == "__main__":
    executor = htf.Test(
        main_phase
    )
    executor.execute()
```
I want to call this explicitly. (My tests are broken up into modules, which I want to import from time to time)

Yes, I am aware that I could describe this in an htf.Test.

My place of concern is that when making, say:
```python
def run_explicit(*fn: List[Callable]):
    _executor = htf.Test(*fn)
    _executor.execute()
```
my plugs get initialized twice, which does not apply to my use case.

dbhatman, 2023-06-05:
I am a little confused about why your plugs are getting initialized twice. Where do you call run_explicit here? Inside main_phase?

I am curious if you could instead make a separate function that does the operations of cool phase as necessary? That could look like this
```
COOL_PLUGS = {plug: Plugger}
COOL_MEASUREMENTS = [...]

def cool_func(api: TestApi, plug: Plugger) -> :
  return doing_cool_stuff(...) == Success

@htf.plugs(**COOL_PLUGS)
@htf.measurets(*cool_measurements)
def cool_phase(api: TestApi, plug: Plugger):
  cool_func(api, plug)
```

```
<other module>

@htf.plugs(other_plug=OtherPlug, **cool_module.COOL_PLUGS)
@htf.measurets(measurement1, measurement2, *cool_module.cool_measurements)
def main_phase(api: TestApi, plug: Plugger):
  cool_module.cool_func(api, plug)
```
Then you would use main_phase vs cool_phase where relevant.

That said you can do this with something like the following:
```
@htf.plugs(stuff=SomePlug)
def cool_phase(api: htf.TestApi, stuff: SomePlug):
    return stuff.do_stuff() == StuffResult.SUCCESS

# Copy plugs and measurements over
# They must get passed in so this test_state (test_state is created when the executor calls main_phase) has them but maybe only use them inside cool_phase.
@htf.plugs(other_plug=OtherPlug, **{plug.name: plug.cls for plug in cool_phase.plugs})
@htf.measures(*cool_phase.measurements)
@htf.PhaseOptions(requires_state=True)
def main_phase(test_state: test_state.TestState, other_plug: OtherPlug, **cool_plugs):
  del cool_plugs
  cool_phase(test_state)
  api_for_other_stuff = test_state.test_api
    
if __name__ == "__main__":
    executor = htf.Test(
        main_phase
    )
    executor.execute()
```
Basically take a look at requires_state: https://github.com/google/openhtf/blob/5f11f03af022f7d041d14f135512ca9e71d8359b/openhtf/core/phase_descriptor.py#L101

kehrazy, 2023-06-05:
Looks promising, will check this out later.

kehrazy, 2023-06-07:
I was being a smartass.

The plugs were initializing twice due to me creating multiple instances of an `openhtf.Test`. You can call these directly, once you're in a PhaseDescriptor of sorts.

closed 2023-06-07 by kehrazy