google/openhtf #1108: runtime reporting/logging for each phase during test

nikitos1550, 2023-07-26:
Hi! 
I am new to OpenHTF and I have question.
How can I make some logging output (actually I want make HTTP POST queries, but it actually does not matter) before and after each phase? 
There is output callback, but output is called after all test (all phases) is done.
Of course I can add some code to each phase in the beginning and end to perform needed actions, but it is not convenient and also, as I can think measurements will are not be checked, need something that will call me before and right after phase with all phase info and results. Are there some mechanics that can allow me do this?

nikitos1550, 2023-07-26:
P.S. I have seen that web server example do some openthf object polling, but I do not think it is proper solution.

kehrazy, 2023-08-12:
As a temporary workaround you could wrap your tests in decorators.

```python
def send_logs(test_fn):
    #: send logs
    yield test_fn
    #: send logs once again

@send_logs
def my_cool_test(...):
```

nikitos1550, 2023-08-15:
> As a temporary workaround you could wrap your tests in decorators.
> 
> ```python
> def send_logs(test_fn):
>     #: send logs
>     yield test_fn
>     #: send logs once again
> 
> @send_logs
> def my_cool_test(...):
> ```

How can I use this? will it be invoked after phase finished or before? how to get logs?

nikitos1550, 2023-08-16:
Seems I uderstand, as temporary solution it is ok, but pain to add it for each test )))

nikitos1550, 2023-08-16:
Are there any plans for openhtf improvements? I mean some new feature to make it more suitable for general testing? Are there any place where I can share my ideas?

nikitos1550, 2023-09-02:
After some source code reading, I realize that proper way is Station API, something like my own Station Watcher.
So answer for my original question is Station API.

closed 2023-09-02 by nikitos1550