google/openhtf #1109: Having trouble using the unit test decorator @test.patch_plugs to mock a device I'm testing.

mikefhill, 2023-08-14:
Hello! 

I'm trying to write unit tests for my test phases with measurements from a mocked plug. I've followed the examples provided in openhtf/openhtf/util/test.py, but keep getting the following error each time I run the unit test: 

openhtf.util.test.InvalidTestError: ('Methods decorated with patch_plugs or yields_phases must yield test phases or openhtf.Test objects.', None)

Here is an example of how I'm using the mock plug:
`  @test.yields_phases
    @test.patch_plugs(mock_device=test_module.DevicePlug)
    def test_calibrate_sst(self, mock_device):
        mock_device.calibrate_sst.return_value = True
        test_rec = yield htf.Test(mock_device.calibrate_sst)
        self.assertTestPass(test_rec)`

Any help or insight on how to fix this would be greatly appreciated, thanks!

glados-verma, 2023-08-14:
1. Make sure your test fixture derives from `test.TestCase` (imported from openhtf) and not (just) the Python standard `unittest.TestCase`.
2. Remove the `@test.yields_phases`, not necessary if you're using `patch_plugs`

Beyond that, code looks fine? Although it's not clear why you're passing `mock_device.calibrate_sst` as an argument to `htf.Test`. You should be passing the phases to test (is `calibrate_sst` a phase defined on the `DevicePlug`?)

You could also try using `TestCase.execute_phase_or_test` and skip the `patch_plugs` decorator, might be easier to write tests that way. The module docs have examples on how to use this function.

mikefhill, 2023-08-16:
Using `TestCase.execute_phase_or_test` did the trick! Thanks for your help!

glados-verma, 2023-08-16:
Excellent! I do recommend using `execute_phase_or_test` over the older `@patch_plugs` decorator.

closed 2023-08-16 by glados-verma