google/openhtf #1136: Status LED on failure

sibertdeclercq, 2024-02-16:
One of the plugs I have written is an RGB LED to indicate different phases and statusses with. The problem I'm running into is that the test result is only available after the `test.execute` as far as I can see.

This means I don't have my plugs available to set the LED to red. I'm now abusing `teardown` to read the test record outcome to:
1. Set the LED
2. Wait for my DUT to detach to start the next cycle

Is there a better wat to achieve this? I've looked into output callbacks but I couldn't find if it's possible to pass plugs to the callbacks.

Some pointers would be appreciated!

mbenabda, 2024-03-27:
you could write a client to the websocket server that the station and dashboard servers expose.

a quick and dirty implementation:
```python
import collections
import itertools
import json
import logging
import asyncio
import secrets
import string
import time
from typing import Union
import websockets
import random
from gpiozero import RGBLED
import backoff
import argparse

RED    = (1, 0, 0)
GREEN  = (0, 1, 0)
BLUE   = (0, 0, 1)
PURPLE = (1, 0, 1)

KEEP_LED_IN_CURRENT_STATE = lambda: None

logger = logging.getLogger()

ONE_MINUTE = 60
def unwrap(payload):
  payload = str(payload)
  return payload[1:] if len(payload) > 1 else None

@backoff.on_exception(backoff.expo, ConnectionRefusedError, max_time=ONE_MINUTE)
@backoff.on_exception(backoff.expo, websockets.ConnectionClosed, max_time=1)
async def test_status_ui(logger, uri, station_id, led):
  async with websockets.connect(uri) as websocket:
    logger.info(f"Started test status UI")

    while True:
      payload = await websocket.recv()
      if payload is None:
        break # TODO continue instead ?
      try:
        msg = unwrap(payload)

        if msg is None:
          continue

        latest_test = latest_test_execution(json.loads(msg), station_id)
        show_test_status(logger, latest_test, led)
      except:
        logger.exception(f"unable to handle payload: {payload}")
        continue # ignore

@backoff.on_exception(backoff.expo, ConnectionRefusedError, max_time=ONE_MINUTE)
@backoff.on_exception(backoff.expo, websockets.ConnectionClosed, max_time=1)
async def station_status_ui(logger, uri, station_id, led):
  was_connected = False
  try:
    async with websockets.connect(uri) as websocket:
      logger.info(f"Started station UI")

      was_connected = True
      while True:
        payload = await websocket.recv()
        if payload is None:
          break # TODO continue instead ?
        try:
          msg = unwrap(payload)
[... 137 lines cut ...]
        red           = config.test_status_red_led_gpio,
        green         = config.test_status_green_led_gpio,
        blue          = config.test_status_blue_led_gpio,
        initial_value = (0, 0, 0),
        active_high   = True,
        pwm           = False
      )
    )
  ]

  logger.info(f"Starting UI for station of id '{config.station_id}' at {config.dashboard_server_host}:{config.dashboard_server_port}")

  loop = asyncio.get_event_loop()
  done, pending = loop.run_until_complete(asyncio.wait(
    [ asyncio.ensure_future(ui) for ui in uis ],
    return_when = asyncio.FIRST_COMPLETED
  ))

  for task in pending:
    task.cancel()

  loop.close()

  exit(len(pending))
```

jpl-harmattan, 2025-03-26:
Hey @sibertdeclercq 

I'm pretty new to OpenHTF so my idea may be completely off, but I think you can store data about the ongoing test in the `test.state`attribute. That way you could change the `test.state` in one phase, and then reuse that information in following phases? This could allow you to set the LED accordingly without needing to wait for the whole test to execute.

From my understanding `test_state` is initialized as an empty dict, so you can populate it just like a regular dictionary with a (key, value) pair:  `test.state[key] = value`

After re-reading your question, I can also interpret it as though you would like to change the LED color depending on the Phase outcome (pass, fail, repeat...) and not just the entire Test outcome. In that case, you might want to make a call to your plug in each of your phases and then update the LED color when you define the phase outcome (`openhtf.PhaseResult.CONTINUE` or other outcome).

My guess is I'll probably end up having a similar use case as yours once I get into some more in depth testing so I'm glad to try and help you understand how to solve the issue. 

I think it would be helpful if you illustrated your problem with some code or a more detailed explanation.

Best,

Jpl