google/openhtf #1200: Update of install items

didjado, 2024-12-07:
It seams that there is multiple issues with the packages names  two items need the "_" changed to a "-"

And it would be really help full id someone could make a release to pypi

glados-verma, 2024-12-11:
Can you expand on what issues you're seeing? Installation is routinely exercised via GitHub Actions and it doesn't look broken there.

didjado, 2024-12-12:
None of the examples will run they all have path issues or the libraries they try to call on trying to run don't load due to being incorrect names

also im really not sure what version of python i should use.. i was using spintop openhtf from pypi but thats now dead and i need to use a new version of python... but i cannot get any of the exampes to run that have a GUI and they all need me to correct package paths etc

glados-verma, 2024-12-18:
Opened #1206 for the missing protobufs.

Separately I'll update the examples and docs.