google/openhtf #1208: Landing page example broken

manuel-91, 2025-01-09:
Hello,

the very basic example on the landing page looks very good and I wanted to get a feel of it. I'm totally new to openhtf. I had a hard start, because it doesn't run :-/

`___init__` requires a port:

```python
class MultimeterPlug(BasePlug):
   def __init__(self, port):
     ...
```

But I found no way to provide i.e. "COM1" with the decorators, here the usage of that class:
```python
@plug(multimeter=MultimeterPlug)
@htf.measures(...)
def test_voltage(test, multimeter):
  ...
```

I'm getting ```Error: __init__() missing 1 required positional argument: 'port'``` How to fix that?

Here the full code, in one file and with added indentation:
```python
import openhtf as htf
from openhtf.plugs import plug, BasePlug

class MultimeterPlug(BasePlug):
    # Initialize the multimeter plug with a port connection
    def __init__(self, port):
      self.port = port
      self.connection = f"Connected on {self.port}"

    def measure_voltage(self):
      # Logic to simulate a voltage measurement
      return 3.3

    # Automatically called at test teardown to clean up
    def tearDown(self):
      # Logic to close the connection or clean up resources
      self.connection = None

@plug(multimeter=MultimeterPlug)
@htf.measures(
  htf.Measurement('voltage')
  .in_range(3.0, 3.5)
  .with_units('V'))
def test_voltage(test, multimeter):
  voltage = multimeter.measure_voltage()
  test.measurements.voltage = voltage

def main():
  test = htf.Test(test_voltage)
  test.execute(lambda: "PCB0001") # UUT S/N

if __name__ == '__main__':
    main()
```

I tried changing `@plug(multimeter=MultimeterPlug)` to `@plug(multimeter=MultimeterPlug("COM1")` but that's not valid:
`TypeError: issubclass() arg 1 must be a class`

The library expects a class and not an instance, makes sense. But how to fix that? Am I overseeing something? A quickfix would be to hardcode the port. But it would be nice if the port could be defined from the testcode without a need of an extra wrapper or hardcoding. Would be great if such an awesome library would have a working example on the landing page... or if someone help me seeing my issue!

glados-verma, 2025-01-09:
A ha, the landing page you speak of is a website that isn't ours. You can reach the owners of that website regarding their content.

closed 2025-01-09 by glados-verma

manuel-91, 2025-01-10:
Thank you for the quick reply! You are correct here https://github.com/google/openhtf is no example.   With landingpage i meant https://www.openhtf.com/ . Is this issue tracker the wrong place to ask/complain?

upview, 2025-01-14:
Hi,

I'm Julien from TofuPilot. We're the creators and maintainers of the [OpenHTF Documentation](https://openhtf.org/).

Thanks for reporting the issue, and sorry for the inconvenience!

We've fixed the homepage by simplifying the example and added an advanced use case here: https://www.openhtf.com/plugs#advanced-use-cases to show multiple plug configurations.

Let me know if anything else is missing!

Best,
Julien

kbroch-rivosinc, 2025-01-14:
Here's an open issue for documentation for this project: https://github.com/google/openhtf/issues/44

I also find it confusing the only usable documentation is provided by a third party SAAS company. 

@upview : Is that documentation open source?  I don't find it here: https://github.com/tofupilot. Why couldn't the non-tofupilot documentation be contributed to this project?

upview, 2025-01-15:
Hi @kbroch-rivosinc!

You're right, it should be open-source. We're planning it for the next sprint.

It started as a guide for TofuPilot users with OpenHTF, but now that it's useful to others, it makes sense to share it with the community.

glados-verma, 2025-01-15:
Issue #44  has been open and "help wanted" for a while now. We will happily review any documentation PRs!

I've just created an empty wiki, please send over your documentation PRs to that.

upview, 2025-02-10:
We made the openhtf.com documentation public here: https://github.com/tofupilot/openhtf-docs