google/openhtf #1219: Standalone executable w. PyInstaller

aunsbjerg, 2025-03-07:
I have a small application based on openhtf that I want to distribute to some non-technical people in my company, so to makes things easier, providing a single executable is preferable. Currently, I am looking into pyinstaller, but having some issues.

Essentially, I am able to create an executable in pyinstaller with `pyinstaller --collect-data=openhtf --paths=production-tester --onedir production-tester/tester.py`, but when running the executable, I get the following exception:

```sh
❯ ./dist/tester/tester --help
Traceback (most recent call last):
  File "tester.py", line 11, in <module>
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "PyInstaller/loader/pyimod02_importers.py", line 450, in exec_module
  File "targets/chamber.py", line 68, in <module>
  File "openhtf/plugs/__init__.py", line 110, in result
    phase = phase_descriptor.PhaseDescriptor.wrap_or_copy(func)
  File "openhtf/core/phase_descriptor.py", line 259, in wrap_or_copy
    retval = cls(func)
  File "<attrs generated init openhtf.core.phase_descriptor.PhaseDescriptor>", line 6, in __init__
    self.func_location = __attr_factory_func_location(self)
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "openhtf/core/phase_descriptor.py", line 219, in _func_location
    line_number = inspect.getsourcelines(obj)[1]
                  ~~~~~~~~~~~~~~~~~~~~~~^^^^^
  File "inspect.py", line 1238, in getsourcelines
  File "inspect.py", line 1078, in findsource
OSError: could not get source code
[PYI-2911318:ERROR] Failed to execute script 'tester' due to unhandled exception!
```

It seems that openhtf does some runtime introspection that isn't "captured" by pyinstaller.

I realize that this might not be a openhtf issue as much as it is a pyinstaller issue, but I am asking in the hope that someone else might have had the same problem and found a solution.

glados-verma, 2025-03-07:
If you examine the `tester` directory created by Pyinstaller, do you see all your test dependencies in there? Source files with phase definitions and plug definitions in particular.

I'm not familiar with Pyinstaller, but we can start by checking whether it's bundling the files you need.

aunsbjerg, 2025-03-10:
> If you examine the `tester` directory created by Pyinstaller, do you see all your test dependencies in there? Source files with phase and plug definitions in particular.
> 
> Are you using the `openhtf.util.argv` or `configuration` module at all? The `argv` module uses `__import__` which pyinstaller won't be able to track. Again, check if you're missing any files in the Pyinstaller-created directory.
> 
> I'm not familiar with Pyinstaller, but we can start by checking whether it's bundling the files you need.

You lead me on the right path; I did some digging into what actually happens and it turns out there is a couple of things going on. The first problem was my own project files, specifically a module that implements some test cases with the openhtf decorators. Openhtf does introspection to find the source location of the test case, so that failed because my project files where not added. Adding the project files to the pyinstaller executable as data fixed that issue.

Another issue is that openhtf does introspection of it's own source in some cases. I believe it was the user_input module, but I forget. That can be fixed by adding openhtf python sources as data to the executable. I added a hook to ensure openhtf python sources are bundled as data files by pyinstaller, placed in `hooks/hook-openhtf.py`:

```python
from PyInstaller.utils.hooks import collect_data_files

datas = collect_data_files("openhtf", include_py_files=True)
```

I had some other issues not related to openhtf related to my project structure, but that was fixed by adding a hook to add the source folder to the pythonpath at runtime.

So finally, I can create a single executable file containing openhtf, other dependencies and my own source like so:

```sh
pyinstaller \
  --noconfirm \
  --onefile \
  --additional-hooks-dir=hooks \
  --add-data "production-tester:production-tester" \
  --paths=production-tester \
   production-tester/tester.py
```

So, I don't believe openhtf does anything "wrong", but it also does not work out of the box because of the introspection. Pyinstaller is perfectly able to deal with it but with a bix of tweaking.

glados-verma, 2025-03-11:
That's very useful to know, thanks for writing it up! I'll move this bug to the "documentation" component, something to mention in the Wiki when we get down to populating it.